STMicroelectronics/stm32h5xx-hal-driver #3: Missing pre-processor directive "defined" cause warning.

42Bastian, 2024-02-19:
Commit: b21b55fe8b05245d06c559a09a77edf5a0679a74
File: stm32h5xx_hal_def.h
Line: 178

`#if defined ( __CC_ARM   ) || ((__ARMCC_VERSION) && (__ARMCC_VERSION >= ARMCC_MIN_VERSION))`

=>

`#if defined ( __CC_ARM   ) || (defined (__ARMCC_VERSION) && (__ARMCC_VERSION >= ARMCC_MIN_VERSION))`

TOUNSTM, 2024-02-22:
Hello @42Bastian,

Thank you for your contribution. Could you please confirm if the issue occurs solely when the `-Wundef` flag is activated? 

With regards,

42Bastian, 2024-02-22:
Only with -Wall (whicht includes -Wundef).

TOUNSTM, 2024-03-07:
See also https://github.com/STMicroelectronics/STM32CubeU5/issues/16

42Bastian, 2024-03-07:
Wow, I forgot about this. And in two years no fix?! :(

TOUNSTM, 2024-05-09:
ST Internal Reference: 181063

ra1nb0w, 2024-07-02:
obviously, same issue for STM32CubeH5

ALABSTM, 2024-11-25:
> obviously, same issue for STM32CubeH5

Hi @ra1nb0w,

The STM32CubeH5 repository points to this one using the "submodule" technique. Once the link updated, the fix will propagate to that repository too.

With regards,

closed 2024-11-25 by ALABSTM