STMicroelectronics/stm32h5xx-hal-driver #4: HAL_ADC_Start_DMA gets too slow if temp. channel is used

koniarik, 2024-03-10:
**Describe the set-up**

 * Custom board, using STM32H503CBTx
 * arm-none-eabi-gcc (Arch Repository) 13.2.0

**Describe the bug (skip if none)**

`HAL_ADC_Start_DMA` get's really slow when internal temp sensor is used on channel that is not strictly related to temperature.

**How to reproduce the bug (skip if none)**

Setup round-robin setup for ADC channels, which at 10kHz does one of the following:
 1) Chann0 is read by DMA for multiple values
 2) Internal temp. is read once

In step 1), usage of `HAL_ADC_Start_DMA` function is really slow.

The problem disappears if internal temp. channel is not used at all.

**Additional context**

Based on reading of the source code, the problematic place is this:
https://github.com/STMicroelectronics/stm32h5xx_hal_driver/blob/main/Src/stm32h5xx_hal_adc.c#L3519

ADC_Enable will execute the "sensor stabilization delay", which takes a considerable amount of time. (Relative to it being run in 10kHz loop) This seems as waste of time given that the channel being started is NOT the internal temperature channel

RJMSTM, 2024-03-14:
Hello @koniarik,

The ADC has been configured to perform two tasks in a 10KHz loop. Firstly, it reads channel 0 using DMA for multiple reads, and secondly, it reads the temperature sensor. This means that the same ADC is being utilized for multiple tasks and is reconfigured each time.
An issue arises when the temperature sensor channel remains active, causing a delay in the stabilization process of the temperature sensor. This delay further causes the ADC to wait when it needs to read channel 0.

To optimize the process, my recommendation would be to disable the temperature sensor when it is not in use. This will prevent any unnecessary delays and ensure that time is not wasted.

Thank you again for your contribution.
With regards,

koniarik, 2024-03-14:
How? And frankly, is it trully the best idea?

My thouhts:
 - The thing is that the this behavior is not exactly clear from the API nor the documentation in adc.c file  (and something I would prefer to be warned about)
 - Suddenly it's not clear to me what is the intended usage of the peripheral... what is the proper way to use it? or what is the proper way to disable just the temperature sensor?
 - Can I do the disable any other way than disabling entire peripheral and enabling again?

koniarik, 2024-03-15:
I tried to apply the fix but it does not work

I don't have time to investigate this now, as based on my reading of the code it should've worked, I will answer here once I have time for deeper investigation :)

RJMSTM, 2024-03-21:
Hello,
Can you please find the time to test the proposed solution?

I appreciate your honesty and look forward to hearing back from you once you have had the time to conduct a more thorough investigation. In the meantime, please let me know if there is anything I can do to assist you.

Regards, 
Rania

koniarik, 2024-03-21:
Sorry for the way!

Turns out that fix suggested by @ARIOSTM works, there was failure on my side in evaluation.

The proposed fix indeed gets rid of the waiting when different channel is used. It's just that even the wait for temperature channel only switch only, the system is too unresponsive and triggers our internal warnings. But I don't blame that on the HAL, as that is just a property of the temp. sensor.

Thanks!

RJMSTM, 2024-03-29:
Hello @koniarik,

Thank you for your feedback, If the solution has resolved your problem, so allow me to close the issue.
We're delighted to hear from you once more.

Regards,
Rania

closed 2024-04-05 by ALABSTM