STMicroelectronics/stm32h5xx-hal-driver #6: The constant "LL_RCC_USB_CLKSOURCE_PLL2Q" is missing from file "stm32h5xx_ll_rcc.h"

iizonn, 2024-05-30:
File: **Inc/stm32h5xx_ll_rcc.h** (sorry, I'm new to the "issue" section of github and not sure how to find or embed the commit number.)

Lines starting at: 918

The constant: **LL_RCC_USB_CLKSOURCE_PLL2Q** is missing from the file.

If you use the IDE to set the USB clock source to PLL2Q you will get an undeclared error. The work around is to add:

> #ifndef LL_RCC_USB_CLKSOURCE_PLL2Q
> #define LL_RCC_USB_CLKSOURCE_PLL2Q   RCC_CCIPR4_USBSEL_1       /\*!< PLL2 Q clock used as USB clock source \*/
> #endif

to the "main.h" file.

My best guess as to what is needed to fix the issue, is the lines:

> 918 #define LL_RCC_USB_CLKSOURCE_NONE        0x00000000U               /\*!< No clock used as USB clock source \*/
> 919 #define LL_RCC_USB_CLKSOURCE_PLL1Q       RCC_CCIPR4_USBSEL_0       /\*!< PLL1 Q clock used as USB clock source \*/
> 920 #if defined(RCC_CR_PLL3ON)
> 921 #define LL_RCC_USB_CLKSOURCE_PLL3Q       RCC_CCIPR4_USBSEL_1       /\*!< PLL3 Q clock used as USB clock source \*/
> 922 #endif /\* PLL3 \*/
> 923 #define LL_RCC_USB_CLKSOURCE_HSI48       RCC_CCIPR4_USBSEL         /\*!< HSI48 clock used as USB clock source \*/

should be something like:

> 918 #define LL_RCC_USB_CLKSOURCE_NONE        0x00000000U               /\*!< No clock used as USB clock source \*/
> 919 #define LL_RCC_USB_CLKSOURCE_PLL1Q       RCC_CCIPR4_USBSEL_0       /\*!< PLL1 Q clock used as USB clock source \*/
> 920 #if defined(RCC_CR_PLL3ON)
> 921 #define LL_RCC_USB_CLKSOURCE_PLL3Q       RCC_CCIPR4_USBSEL_1       /\*!< PLL3 Q clock used as USB clock source \*/
> 922 #else
> 923 #define LL_RCC_USB_CLKSOURCE_PLL2Q       RCC_CCIPR4_USBSEL_1       /\*!< PLL2 Q clock used as USB clock source \*/
> 924 #endif /\* PLL3 \*/
> 925 #define LL_RCC_USB_CLKSOURCE_HSI48       RCC_CCIPR4_USBSEL         /\*!< HSI48 clock used as USB clock source \*/

ALABSTM, 2024-06-03:
Hi @iizonn,

Issue reproduced using a `STM32CubeMX` generated example. Thank you for the fix proposal. Looks quite relevant. This will be forwarded to our development team to have it fixed. I will keep you informed.

With regards,

ALABSTM, 2024-06-04:
ST Internal Reference: 183323

closed 2024-11-25 by ALABSTM