STMicroelectronics/stm32h5xx-hal-driver #7: ETH Rx Tailpointer calculation wrong

gresolio, 2024-08-15:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post your report to the **ST Community** in the STM32 MCUs dedicated [page](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus).

**Describe the set-up**
[NUCLEO-H563ZI](https://www.st.com/en/evaluation-tools/nucleo-h563zi.html) with STM32H563ZIT6

**Describe the bug**
In the function ETH_UpdateDescriptor the calculation of the tailidx/DMACRDTPR is wrong.
Because of this the HAL_ETH_ErrorCallback is reached in my application.
The tail pointer should point to the last built/updated descriptor.

Current (v1.3.0) implementation:
https://github.com/STMicroelectronics/stm32h5xx_hal_driver/blob/0de1dbf8ad03e6e466ed55d1c174656561e3240d/Src/stm32h5xx_hal_eth.c#L1212

Solution to fix the tail pointer:
```c
tailidx = (ETH_RX_DESC_CNT + descidx - 1U) % ETH_RX_DESC_CNT;
```

The related issue in the stm32h7xx_hal_driver reported by Florian-Fischer-InMach:
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/issues/61

ASEHSTM, 2024-08-16:
Hello @gresolio,

Thank you for your contribution. You are right, this will be fixed soon.
ST Internal Reference: 188836

With Regards,

ASEHSTM, 2024-09-05:
Fixed in 7e8c7d8e91c15321864f513f8458ce7f6555eb4e

closed 2024-09-05 by ASEHSTM