STMicroelectronics/stm32h5xx-hal-driver #8: HAL_SPI_Abort_IT does not work for bidirectional DMA transfers due to incorrect logic

multiplemonomials, 2024-09-13:
**Caution**

The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post your report to the **ST Community** in the STM32 MCUs dedicated [page](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus).

**Describe the set-up**

Board: Nucleo H563ZI
Compiler: arm-none-eabi-gcc 13.2.1

**Describe the bug (skip if none)**

Doing some testing, I noticed that when I aborted a bidirectional DMA SPI transfer with HAL_SPI_Abort_IT, the Rx DMA channel's state (`hspi->hdmarx->State`) would not go back to READY.  This caused the next transfer to spuriously fail.

Why did it not go back to READY?  It turns out that [this line](https://github.com/STMicroelectronics/stm32h5xx_hal_driver/blob/7e8c7d8e91c15321864f513f8458ce7f6555eb4e/Src/stm32h5xx_hal_spi.c#L2631) causes SPI_DMATxAbortCallback() to be called, which in turn calls SPI_AbortTransfer().  That function [clears both the RXDMAEN and TXDMAEN bits](https://github.com/STMicroelectronics/stm32h5xx_hal_driver/blob/7e8c7d8e91c15321864f513f8458ce7f6555eb4e/Src/stm32h5xx_hal_spi.c#L3627).  Then, back in HAL_SPI_Abort_IT(), we check the RXDMAEN bit to determine if we need to abort the Rx DMA.  However, that bit was already cleared earlier, so [this if statement](https://github.com/STMicroelectronics/stm32h5xx_hal_driver/blob/7e8c7d8e91c15321864f513f8458ce7f6555eb4e/Src/stm32h5xx_hal_spi.c#L2648) is never entered and we don't abort the Rx DMA.

**How to reproduce the bug (skip if none)**

1. Do any long-duration, bidirectional DMA SPI transfer
2. Call HAL_SPI_Abort_IT()
3. Observe that the Rx DMA state is not READY after the abort function returns and that the next transfer, if attempted, will fail due to this

**Additional context**

Fixing this is pretty easy.  We just have to save the value of the Rx DMA flag before the Tx DMA abort.  Like this:
```cpp
  // Save Rx DMA flag as this gets cleared by SPI_AbortTransfer()
  int rxDMAUsed = HAL_IS_BIT_SET(hspi->Instance->CFG1, SPI_CFG1_RXDMAEN);
  if (hspi->hdmatx != NULL)
  {
    if (HAL_IS_BIT_SET(hspi->Instance->CFG1, SPI_CFG1_TXDMAEN))
    {
<snip>
    }
  }

  if (hspi->hdmarx != NULL)
  {
    if (rxDMAUsed) // <-- Use this instead of checking the RXDMAEN bit
    {
```

KRASTM, 2024-09-16:
ST Internal Reference: 191186

KRASTM, 2024-10-11:
Fixed in commit https://github.com/STMicroelectronics/stm32h5xx_hal_driver/commit/fcd3e48929ec60600e18df19a1cd12d214ebdc6d

closed 2024-10-11 by KRASTM