STMicroelectronics/stm32h5xx-hal-driver #9: stm32h5xx_ll_i3c.c doesn't compile when both USE_HAL_DRIVER and USE_FULL_LL_DRIVER are defined

Aric5301, 2024-09-29:
**Describe the set-up**

 * The board: Nucleo-H503RB
 * Compiler: arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi

**Describe the bug (skip if none)**

When both USE_HAL_DRIVER and USE_FULL_LL_DRIVER are defined, stm32h5xx_ll_i3c.c doesn't compile.
The produced error is as follows:

_In file included from Inc/stm32h5xx_hal_conf.h:344,
                 from Inc/stm32h5xx_hal.h:29,
                 from Device/ST/STM32H5xx/Include/stm32h5xx.h:230,
                 from Inc/stm32h5xx_ll_i3c.h:28,
                 from Src/stm32h5xx_ll_i3c.c:27:
Inc/stm32h5xx_hal_i3c.h:50:3: error: unknown type name 'LL_I3C_CtrlBusConfTypeDef'
   _50 |   LL_I3C_CtrlBusConfTypeDef CtrlBusCharacteristic; /*!< Specifies the I3C controller bus characteristic configuration
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~
Inc/stm32h5xx_hal_i3c.h:53:3: error: unknown type name 'LL_I3C_TgtBusConfTypeDef'
   53 |   LL_I3C_TgtBusConfTypeDef  TgtBusCharacteristic;  /*!< Specifies the I3C target bus characteristic configuration
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
Inc/stm32h5xx_hal_i3c.h:1137:62: error: unknown type name 'LL_I3C_CtrlBusConfTypeDef'
 1137 |                                                        const LL_I3C_CtrlBusConfTypeDef *pConfig);
      |                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~
Inc/stm32h5xx_hal_i3c.h:1139:61: error: unknown type name 'LL_I3C_TgtBusConfTypeDef'
 1139 |                                                       const LL_I3C_TgtBusConfTypeDef *pConfig);
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~__

**How to reproduce the bug**

Here's the full compilation command I use for your convenience:
~/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gcc Src/stm32h5xx_ll_i3c.c -o main.o -Wall -Wextra --specs=nosys.specs -std=c17 -mcpu=cortex-m33 -isystem Inc -isystem Inc/Legacy -isystem Include -isystem Device/ST/STM32H5xx/Include -DUSE_FULL_LL_DRIVER -DSTM32H503xx

I implemented stubs for system calls, and added to the include path, in addition to the HAL, the include folders of CMSIS-Core and of the STM32H5xx device.

**Additional context**

After some investigation, it looks to be a problem with recursive includes.
stm32h5xx_ll_i3c.c ->(includes) **stm32h5xx_ll_i3c.h** -> stm32h5xx.h -> stm32h5xx_hal.h -> stm32h5xx_hal_conf.h -> stm32h5xx_hal_i3c.h -> **stm32h5xx_ll_i3c.h**.

The nested stm32h5xx_ll_i3c.h include is then guarded from being included a second time, while the earlier included stm32h5xx_ll_i3c.h has included stm32h5xx.h before defining the needed  types. Therefore, the including stm32h5xx_hal_i3c.h has unknown type names.

ALABSTM, 2024-10-02:
Hi @Aric5301,

Thank you for this report. Issue confirmed. We'll dig deeper into it and get back to you as soon as we get a better insight .

With regards,

ALABSTM, 2024-10-03:
Hi @Aric5301,

I hope you are fine. Looks like there is a misuse of the `I3C` drivers. For a particular peripheral, `I3C` in our case, **whether** we use the `LL` driver or the `HAL`, but **not both at the same time**.

In your case, I have understood that you are using the `LL` `I3C` driver. Hence, in your `stm32h5xx_hal_conf.h`, the line below should be commented.

https://github.com/STMicroelectronics/stm32h5xx_hal_driver/blob/7e8c7d8e91c15321864f513f8458ce7f6555eb4e/Inc/stm32h5xx_hal_conf_template.h#L60

I hope this makes things clearer. Please allow me to close this issue.

With regards,

closed 2024-10-03 by ALABSTM

Aric5301, 2024-10-03:
Hey @ALABSTM, thanks for the reply!
I would like to raise another point. It's not the first `STM32` `HAL` that I'm working with, and for previous `HAL`s I've always used either `LL `or `HAL` for each peripheral, but never have I had a problem of compiling what I don't use. I've always compiled the entire library.
To my understanding, that what you are telling here is that if I would like to use the **`HAL`** `I3C` driver I can't compile the `LL` file.
This is of course acceptable, but it's the first time that I encounter something like that.
Perhaps it should explicitly say somewhere in the manual that for each peripheral that I use, I should either compile the `LL` and disable the peripheral in `..._hal_conf.h` file (and not compile the `HAL` file?), or enable the `HAL`, compile it, and not compile the `LL`. As, again, this unable-to-even-compile behavior is different from what I've encountered up until today.
Thank you!