STMicroelectronics/stm32h5xx-hal-driver #10: Wrong RCC_PERIPHCLK_RTC setup in HAL

jiri-novotny, 2024-12-04:
**Describe the set-up**

 * NUCLEO-H563ZI
 * VSC + CLT 16

**Describe the bug (skip if none)**

Wrong condition in [stm32h5xx_hal_rcc_ex.c](https://github.com/STMicroelectronics/stm32h5xx-hal-driver/blob/main/Src/stm32h5xx_hal_rcc_ex.c#L1743)
exact match of `tmpregister` is not possible with HSE prescaler

**How to reproduce the bug (skip if none)**

* enable RTC clock
```
    PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_RTC;
    PeriphClkInitStruct.RTCClockSelection = RCC_RTCCLKSOURCE_HSE_DIV20;
    if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK)
```
* read backup register
* set backup register
* reset board

backup-domain goes always through reset

ALABSTM, 2024-12-09:
Hi @jiri-novotny,

I see you are using a **NUCLEO-H563ZI** board.
* The default configuration for the HSE clock is having it provided the **ST-LINK**. In this case, your application should use `RCC_HSE_BYPASS` as configuration for clock source instead of `RCC_HSE_ON`. Pay attention this HSE clock frequency is **8MHz** in this case.
* Otherwise, you have to remove the solder bridge **SB49** to be able to use the **25MHz** external crystal X3 as HSE clock.

Please refer to page **19** or to page **26** of the user manual [UM3115](https://www.st.com/resource/en/user_manual/um3115-stm32h5-nucleo144-board-mb1404-stmicroelectronics.pdf) Rev. 2 for more details.

With regards,

ALABSTM, 2024-12-18:
ST Internal Reference: 199007

closed 2025-02-07 by ALABSTM

jiri-novotny, 2025-03-03:
FYI @ALABSTM change is discarded on 1.5.0